Dilbert66/esphome-dsckeybus #179: Compilation Error ESP8266

nicocalvagna, 2025-08-25:
Hello, first of all, I wanted to thank you for this contribution. I've been using this integration with ESPHOME and HA for 3 years, and it's excellent.

I've been having trouble compiling for a few days. I'm using a board with a WeMos mini D1 (ESP8266), and I'm getting the following error. I'm not a programming expert, but with a little research, I came to the conclusion that this is the reason:

Your YAML is fine. The problem comes from the external component (dsc_alarm_panel), which forces SET_LOOP_TASK_STACK_SIZE, even on the ESP8266.

I've been having problems with systematic data bus disconnections. It always reconnects after 15/20 seconds, but it happens after a disarm. It's also failing to arm while present, then disarms OK, and when I re-arm, it no longer works because the system remains in the "arming" state and doesn't update. I deduce that this is happening because the system did send the command on the data bus, but the ESP was rebooting.

I clarify that both physical keyboards are working fine and responding normally.

I wanted to update as soon as possible in case it's a known issue, and perhaps it will be fixed, but I'm running into this problem and can't compile anymore.

Thank you very much in advance.

Greetings from Argentina.

Dilbert66, 2025-08-25:
This has been corrected but you might want to migrate to an esp32 as the new code will not work correctly on an esp8266 due to the recent esphome changes.  The esp8266 is too limited in resources and will keep crashing.

nicocalvagna, 2025-08-26:
First of all, thank you very much for the response. Let's take it step by step.

After the correction, the code compiled correctly on the ESP8266, but unfortunately, it restarts periodically, especially when reading data from the keybus.

Following your advice, I rebuilt the board with an ESP32. Now I'm facing a new problem. The code compiled fine, I can see the status of my system perfectly, and so far, it hasn't restarted.

When I created the device in ESPHome, I used a different name and added it to HA. This caused me the problem of all the entities changing their names. Well, one by one, I changed them, and it seems to be fine. The keypad card looks OK.

The problem is that I don't have any services available in Actions. I assume it's because of the name change, but I have no idea how to find it or fix it.

I tried all the basics: rebooting, recompiling, deleting, and adding again, without success.

I can read all the statuses, but I can't interact because the service doesn't exist. I've already checked the hardware and it's OK. It's not a response error, it's just that it's not there.

Any ideas?

Dilbert66, 2025-08-26:
Ensure you have the custom_services line in the api section.
```
api:
   encryption:
      key: !secret encryption_key
   custom_services: true
```

closed 2025-08-30 by nicocalvagna