Dilbert66/esphome-dsckeybus #180: Sensors not updating after upgrading

abishur, 2025-08-26:
A couple days ago I updated my esp32 after updating esphome to  ESPHome 2025.8.0.  As a part of this my locally stored component files were outdated so I went ahead and switched to using external components and did a clean build however after doing this none of my wired door sensors or motion sensor is updating.

I can still toggle a PGM I have on a GPIO of the ESP32, but there seems to be some sort of break between my esp32 and my dsc panel.  When using the dev branch of the external components, and I try to arm the system I get a Sysem not available feedback

```
23:16:59][I][Panel :968]: [2025-08-25 23:16] F5: F5 00 AF F7 BF FF FB EF DF FF 01 00 00 00 00 00 (74)
[23:16:59][I][dscalarm:807]: Writing keys: N to partition 1, partition disabled: 0 , partition locked: 1
[23:16:59][D][text_sensor:069]: 'line1 partition 1': Sending state 'System'
[23:16:59][D][text_sensor:069]: 'line2 partition 1': Sending state 'not available'
[23:16:59][I][Panel :968]: [2025-08-25 23:16] FD: FD 01 8D FD F7 EF B8 CF FD CF 01 00 00 00 00 00 (74)
[23:16:59][I][Panel :968]: [2025-08-25 23:16] BF: BF 01 FF BD FB EF DF FF FB F7 01 00 00 00 00 00 (74)
```
One thing I do is that I use custom clock and read pins

```
  dscclockpin: "23" #Yellow Wire, usde for SCL
  dscreadpin: "19" #Green Wire, used for SDA
  dscwritepin: "18" #Green Wire, Used for SCK pin to write back (program) to DSC panel
```
Is it possible that somewhere in the external components those got hard coded to the default pins instead of using the custom one I put in?  Or is there some other issue I'm failing to notice?  I made sure to add the "custom_services: true" to the api area.

Dilbert66, 2025-08-26:
Nothing is hard coded. The config file controls the pins .  The partition locked (system not available) means that something is already sendintg to the panel and it can't take another command at that time. I'd have to see more of the log and your config file to see what is going on.  You can try doing a full clean then recompiling from the the esphome dashbaord.

abishur, 2025-08-26:
Did a clean build, still getting the system not ready, here's the log

```
[06:46:55][I][app:200]: ESPHome version 2025.8.1 compiled on Aug 26 2025, 06:46:14
[06:46:55][C][wifi:659]: WiFi:
[06:46:55][C][wifi:442]:   Local MAC: 08:D1:F9:27:ED:10
[06:46:55][C][wifi:447]:   SSID: [redacted]
[06:46:55][C][wifi:450]:   IP Address: 192.168.15.198
[06:46:55][C][wifi:454]:   BSSID: [redacted]
[06:46:55][C][wifi:454]:   Hostname: 'dscalarm'
[06:46:55][C][wifi:454]:   Signal strength: -52 dB ▂▄▆█
[06:46:55][C][wifi:465]:   Channel: 1
[06:46:55][C][wifi:465]:   Subnet: 255.255.254.0
[06:46:55][C][wifi:465]:   Gateway: 192.168.15.1
[06:46:55][C][wifi:465]:   DNS1: 192.168.15.26
[06:46:55][C][wifi:465]:   DNS2: 192.168.15.21
[06:46:55][I][Panel :968]: [2025-08-26 06:46] 3F: 3F 00 EB FF FF FF 7B E7 1E DF 01 00 00 00 00 00 (74)
[06:46:55][C][logger:252]: Logger:
[06:46:55][C][logger:252]:   Max Level: DEBUG
[06:46:55][C][logger:252]:   Initial Level: DEBUG
[06:46:55][C][logger:258]:   Log Baud Rate: 115200
[06:46:55][C][logger:258]:   Hardware UART: UART0
[06:46:55][C][logger:265]:   Task Log Buffer Size: 768
[06:46:55][I][Panel :968]: [2025-08-26 06:46] 25: 25 00 D1 21 7F EF FF C7 80 EF 01 00 00 00 00 00 (74)
[06:46:55][C][switch.gpio:087]: GPIO Switch 'Power Front Door Tablet'
[06:46:55][C][switch.gpio:087]:   Restore Mode: always OFF
[06:46:55][C][switch.gpio:029]:   Pin: GPIO27
[06:46:55][C][switch.gpio:087]: GPIO Switch 'Power Garage Door Tablet'
[06:46:55][C][switch.gpio:087]:   Restore Mode: always OFF
[06:46:55][C][switch.gpio:029]:   Pin: GPIO25
[06:46:55][C][template.switch:087]: Template Switch 'Connection'
[06:46:55][C][template.switch:087]:   Restore Mode: disabled 
[06:46:55][C][template.switch:095]:   Icon: 'mdi:shield-link-variant'
[06:46:55][C][template.switch:057]:   Optimistic: YES
[06:46:55][I][Panel :968]: [2025-08-26 06:46] FF: FF 01 87 35 FE CF FF E7 3D FF 01 00 00 00 00 00 (74)
[06:46:55][C][restart:087]: Restart Switch 'restart_switch'
[06:46:55][C][restart:087]:   Restore Mode: always OFF
[06:46:55][C][restart:095]:   Icon: 'mdi:restart'
[06:46:55][C][safe_mode.switch:087]: Safe Mode Switch 'Safe Mode'
[06:46:55][C][safe_mode.switch:087]:   Restore Mode: always OFF
[06:46:55][C][safe_mode.switch:095]:   Icon: 'mdi:restart-alert'
[06:46:55][C][sntp:041]: SNTP Time:
[06:46:55][C][sntp:044]:   Server 0: '0.pool.ntp.org'
[06:46:55][C][sntp:044]:   Server 1: '1.pool.ntp.org'
[06:46:55][C][sntp:044]:   Server 2: '2.pool.ntp.org'
[06:46:55][C][esphome.ota:075]: Over-The-Air updates:
[06:46:55][C][esphome.ota:075]:   Address: dscalarm.local:3232
[06:46:55][C][esphome.ota:075]:   Version: 2
[06:46:55][C][esphome.ota:082]:   Password configured
[06:46:55][C][safe_mode:018]: Safe Mode:
[06:46:55][C][safe_mode:019]:   Boot considered successful after 60 seconds
[06:46:55][C][safe_mode:019]:   Invoke after 10 boot attempts
[06:46:55][C][safe_mode:019]:   Remain for 300 seconds
[06:46:55][I][Panel :968]: [2025-08-26 06:46] FF: FF 01 CD 91 EB FF 8A F7 F6 FF 01 00 00 00 00 00 (74)
[06:46:55][C][api:205]: Server:
[06:46:55][C][api:205]:   Address: dscalarm.local:6053
[06:46:55][C][api:210]:   Noise encryption: YES
[06:46:55][C][mdns:124]: mDNS:
[06:46:55][C][mdns:124]:   Hostname: dscalarm
[06:46:55][I][Panel :968]: [2025-08-26 06:46] 7F: 7F 01 F5 A1 0F CF EF EF BF FF 01 00 00 00 00 00 (74)
[06:46:55][I][Panel :968]: [2025-08-26 06:46] FF: FF 01 BD FD F7 EF BF FF FB EF 01 00 00 00 00 00 (74)
[06:46:55][I][Panel :968]: [2025-08-26 06:46] DD: DD 01 DF DF 7E FF F7 DF BE FF 01 00 00 00 00 00 (74)
[06:46:55][I][Panel :968]: [2025-08-26 06:46] FD: FD 01 BD FB EF DF 7E FF F7 DF 01 00 00 00 00 00 (74)
[06:46:56][I][Panel :968]: [2025-08-26 06:46] B5: B5 01 BF BF FD F7 EF FF 7D FF 01 00 00 00 00 00 (74)
[06:46:56][I][Panel :968]: [2025-08-26 06:46] F7: F7 00 FB F7 EF FF 7D FF EF DF 01 00 00 00 00 00 (74)
[06:46:56][I][Panel :968]: [2025-08-26 06:46] 8D: 8D 01 BF BF FD F7 EF FF 7D FF 01 00 00 00 00 00 (74)
[06:46:56][I][Panel :968]: [2025-08-26 06:46] F5: F5 00 FB F7 DF FF FD F7 EF FF 01 00 00 00 00 00 (74)
[06:46:56][I][Panel :968]: [2025-08-26 06:46] 7F: 7F 00 A7 FF FB F7 DF FF FD F7 01 00 00 00 00 00 (74)
[06:46:56][I][Panel :968]: [2025-08-26 06:46] E7: E7 00 FF F7 DF FF FD F7 EF FF 01 00 00 00 00 00 (74)
[06:46:56][I][Panel :968]: [2025-08-26 06:46] 7F: 7F 00 8F FF FB F7 DF FF FD F7 01 00 00 00 00 00 (74)
[... 412 lines cut ...]
[06:47:28][I][Panel :968]: [2025-08-26 06:47] AD: AD 01 FF BF 7D FF EF DF 7E FF 01 00 00 00 00 00 (74)
[06:47:28][I][Panel :968]: [2025-08-26 06:47] F5: F5 00 F7 F7 DF FF FD F7 EF FF 01 00 00 00 00 00 (74)
[06:47:28][I][Panel :968]: [2025-08-26 06:47] 57: 57 01 FF 7D FB EF DF FF FB F7 01 00 00 00 00 00 (74)
[06:47:28][I][Panel :968]: [2025-08-26 06:47] EF: EF 00 F7 EF BF FF FB EF DF FF 01 00 00 00 00 00 (74)
[06:47:28][I][Panel :968]: [2025-08-26 06:47] FD: FD 00 9F FD F7 EF BF FF FB EF 01 00 00 00 00 00 (74)
[06:47:28][I][Panel :968]: [2025-08-26 06:47] D5: D5 01 BF BF FD F7 EF FF 7D FF 01 00 00 00 00 00 (74)
[06:47:29][I][Panel :968]: [2025-08-26 06:47] F5: F5 00 F7 F7 DF FF FD F7 EF FF 01 00 00 00 00 00 (74)
[06:47:29][I][Panel :968]: [2025-08-26 06:47] 57: 57 01 BF 7D FB EF DF FF FB F7 01 00 00 00 00 00 (74)
[06:47:29][I][Panel :968]: [2025-08-26 06:47] EF: EF 00 DF EF 7E FF F7 DF BE FF 01 00 00 00 00 00 (74)
[06:47:29][I][Panel :968]: [2025-08-26 06:47] FF: FF 01 BB FB EF DF 7E FF F7 DF 01 00 00 00 00 00 (74)
[... cut ...]
[06:47:29][I][Panel :968]: [2025-08-26 06:47] FF: FF 01 B7 F7 DF FF FD F7 EF FF 01 00 00 00 00 00 (74)
[06:47:30][I][Panel :968]: [2025-08-26 06:47] 5D: 5D 01 BF DF FD F7 EF FF 7D FF 01 00 00 00 00 00 (74)
[06:47:30][I][Panel :968]: [2025-08-26 06:47] F5: F5 00 F7 F7 FF FF FD F7 EF FF 01 00 00 00 00 00 (74)
[06:47:30][I][dscalarm:807]: Writing keys: N to partition 1, partition disabled: 0 , partition locked: 1
[06:47:30][D][text_sensor:069]: 'line1 partition 1': Sending state 'System'
[06:47:30][D][text_sensor:069]: 'line2 partition 1': Sending state 'not available'
```

abishur, 2025-08-26:
And here's the config file

```
#for documentation see project at https://github.com/Dilbert66/esphome-dsckeybus

substitutions:
  # change the name to suit your needs. This is what esphome will use as the friendly name for your component.
  # also used as a prefix for all fields
  name: "dscalarm"
  friendlyName: "DSC Alarm" #friendly name for this device 
  panelId: DSCAlarm  #used as the service variable name.

dsc_alarm_panel:
  id: $panelId
  #you can enter a list of user codes mapped to their names for display in armed/disarmed event messages
  usercodes: "1:Matthew,2:Amy" 
  
  #Only comes into effect if a password prompt occurs when arming eg. night mode
  accesscode: !secret access_code 
  
  #used to select the default partition associated with the alarm panel messages
  defaultpartition: "1" 

  maxzones: "32" # maximum amount of zones your system supports

  # dsc debug level: 0 = off, 1 = minimal, 2 = all packets shown on console  3 =  + serial port debugging
  dscdebuglevel: "3"  
  
  #zone expander addresses:
  # 9  - zones 9-16
  # 10 - zones 17-24
  # 11 - zones 25-32
  # 12 - zones 33-40 (for systems with 64 zone support)
  # 13 - zones 41-48 (for systems with 64 zone support)
  # 14 - zones 49-56 (for systems with 64 zone support)
  # 16 - zones 57-64 (for systems with 64 zone support)  
  expanderaddr1: "9" # 1st zone expander emulator address to use . Set to 0 to disable. 
  expanderaddr2: "10" # 1st zone expander emulator address to use . Set to 0 to disable. 
  
  #ESP32 Pins
  dscclockpin: "23" #Yellow Wire, usde for SCL
  dscreadpin: "19" #Green Wire, used for SDA
  dscwritepin: "18" #Green Wire, Used for SCK pin to write back (program) to DSC panel

  #ESP8266 Pins
  #dscClockPin: "5"
  #dscReadPin: "4"
  #dscWritePin: "15"  

  #periodic check for trouble messages in seconds. Minium 30s. Defaults to 120s (2 minutes)
  trouble_fetch_update_time: 120

  #flag to have the esp check for trouble statuses for devices/zones when the trouble light goes on or off
  #Note: It will prevent the physical panel from beeping to let you know as the check resets the beep.
  #Sends *21 and *27 to the panel. Default: true
  trouble_fetch: true

  #if true the available states will be:
  #triggered,armed_home,armed_away,armed_night,pending,arming,disarmed,ready,not_ready
  #if false the states will be:
  #triggered,armed_home,armed_away,armed_night,pending,arming,disarmed
  #the default is "false"
  detailed_partition_state: true  

  clean_build: "true" #default is false.  set to true if getting duplication errors in linking step. 

  
esp32:
  board: nodemcu-32s
  framework:
[... 339 lines cut ...]
    lambda: |-
      return dsc.keybusConnected;
    icon: "mdi:shield-link-variant"
    turn_on_action:
      - switch.toggle: restart_switch
    turn_off_action:
      - lambda: |-
          disconnectKeybus();

  - platform: restart
    id: restart_switch

  - platform: safe_mode
    name: "Safe Mode" 
    
  - platform: gpio  
    pin: 27
    id: Tab1
    name: Power Front Door Tablet
    
  - platform: gpio  
    pin: 25
    id: Tab2
    name: Power Garage Door Tablet
```

Dilbert66, 2025-08-26:
Ok, a couple things. You can set clean_build to false again. It only needs to be on once to do the clean.  
Secondly, your logs show nonsense so this tells me you have a wiring issue. Perhaps you have your data/clock lines reversed.
Your config looks fine otherwise.

Dilbert66, 2025-08-26:
Well, I'm running it on an esp32 d1 and no issues.  Check that you didnt reverse the clock and data pins on your config.

abishur, 2025-08-26:
I'm pretty sure it's a ESP32 wemos D1 Mini, is there a spot in the logs it would specifically mention that?

I double checked my wiring, I'm definitely using
```
  dscclockpin: "23" #Yellow Wire, usde for SCL
  dscreadpin: "19" #Green Wire, used for SDA
  dscwritepin: "18" #Green Wire, Used for SCK pin to write back (program) to DSC panel
```
When I swapped things around in the code it stopped responding all together.

I also tried making the following change to my config file

```
  #periodic check for trouble messages in seconds. Minium 30s. Defaults to 120s (2 minutes)
  trouble_fetch_update_time: 0

  #flag to have the esp check for trouble statuses for devices/zones when the trouble light goes on or off
  #Note: It will prevent the physical panel from beeping to let you know as the check resets the beep.
  #Sends *21 and *27 to the panel. Default: true
  trouble_fetch: false

  #if true the available states will be:
  #triggered,armed_home,armed_away,armed_night,pending,arming,disarmed,ready,not_ready
  #if false the states will be:
  #triggered,armed_home,armed_away,armed_night,pending,arming,disarmed
  #the default is "false"
  detailed_partition_state: false
```
and it no longer shows system not ready and it doesn't show the partition as being locked anymore, however it's also not letting me set the alarm or showing any response from the motion sensor so I don't think it's actually reading anything, it's just not giving me the error message.

Dilbert66, 2025-08-26:
It's possible that I have a bug in the write line handling since I added duplex capability to it.  I use a level shifter board for my system which only uses 2 lines and duplexes the read/write.  I'll wire up the old still circuit to test that again.   So your system might be fine but the write line might be at the wrong polarity.

Dilbert66, 2025-08-27:
Can you disconnect the dscwritepin (pin12) from the circuit to see if that fixes the logs. I'm looking into the cause but at this time I really can't duplicate the issue on my own systems.

axework, 2025-08-27:
I disconnected the dscwritepin, but unfortunately the status is the same: the temperature seems to update irregulary not every minute as expected. And the DSC sensors seem not to upgrade at all not even the inital state is not correct ( i had the main door open and it is not shown as open).

Here is a part of the log with loglevel 1 to show that at least som temperature readings are coming in every 1 minute.

[17:17:05][D][api.connection:1341]: Home Assistant 2025.8.3 (192.168.1.68) connected
[17:17:11][D][text_sensor:069]: 'Partition 1 Status': Sending state 'disarmed'
[17:17:11][D][text_sensor:069]: 'Trouble Msg': Sending state 'TEL ZT DBAT BAT TAMP SUP RF '
[17:18:00][I][safe_mode:042]: Boot seems successful; resetting boot loop counter
[17:18:00][D][esp32.preferences:142]: Writing 1 items: 0 cached, 1 written, 0 failed
[17:18:02][D][sensor:103]: 'ESP32 Temperature': Sending state 30.60000 °C with 1 decimals of accuracy
[17:19:02][D][sensor:103]: 'ESP32 Temperature': Sending state 30.60000 °C with 1 decimals of accuracy
[17:20:02][D][sensor:103]: 'ESP32 Temperature': Sending state 30.60000 °C with 1 decimals of accuracy
[17:21:02][D][sensor:103]: 'ESP32 Temperature': Sending state 31.60000 °C with 1 decimals of accuracy
[17:22:02][D][sensor:103]: 'ESP32 Temperature': Sending state 32.60000 °C with 1 decimals of accuracy
[17:22:57][D][text_sensor:069]: 'line2 Partition 1': Sending state '123'
[17:22:57][D][text_sensor:069]: 'Partition 1 Msg': Sending state '7b Unknown'
[17:23:02][D][sensor:103]: 'ESP32 Temperature': Sending state 32.60000 °C with 1 decimals of accuracy

Here is small part of the log with loglevel 2 with first test. The log lines increasing rapidly. Could the ESP be overloaded with the logs/info coming in?

[16:58:49][I][Panel :964]: [2025-08-27 16:58] 05: 05 00 85 19 00 C7 0E FF F3 CF 01 00 00 00 00 00 (74)
[16:58:49][I][dscalarm:2067]: status 19, last status 01,selection 01,partition=1,skip=0,force=0
[16:58:49][D][text_sensor:069]: 'line1 Partition 1': Sending state 'Alarm'
[16:58:49][D][text_sensor:069]: 'line2 Partition 1': Sending state 'occurred'
[16:58:49][I][dscalarm:2067]: status FF, last status F7,selection A5,partition=3,skip=0,force=0
[16:58:49][I][dscalarm:2067]: status CF, last status DF,selection A5,partition=4,skip=0,force=0
[16:58:49][D][text_sensor:069]: 'Partition 1 Msg': Sending state '19 Unknown'
[16:58:49][D][binary_sensor:026]: 'Partition 1 Ready': New state is OFF
[16:58:49][I][Panel :964]: [2025-08-27 16:58] 05: 05 00 85 01 00 FF E7 DF 79 E7 01 00 00 00 00 00 (74)
[16:58:49][I][dscalarm:2067]: status 01, last status 19,selection 01,partition=1,skip=0,force=0
[16:58:49][D][text_sensor:069]: 'line1 Partition 1': Sending state 'System is Ready'
[16:58:49][D][text_sensor:069]: 'line2 Partition 1': Sending state 'Ready to Arm <>'
[16:58:49][I][dscalarm:2067]: status DF, last status FF,selection A5,partition=3,skip=0,force=0
[16:58:49][I][dscalarm:2067]: status E7, last status CF,selection A5,partition=4,skip=0,force=0
[16:58:49][D][text_sensor:069]: 'Partition 1 Msg': Sending state '01 Ready'
[16:58:49][D][binary_sensor:026]: 'Partition 1 Ready': New state is ON
[16:58:49][I][Panel :964]: [2025-08-27 16:58] 45: 45 00 85 01 00 C7 00 D7 CF FF 01 00 00 00 00 00 (74)
[16:58:49][I][Panel :964]: [2025-08-27 16:58] 75: 75 00 E5 CF 79 E7 9E FF E7 DF 01 00 00 00 00 00 (74)
[16:58:49][I][Panel :964]: [2025-08-27 16:58] 7D: 7D 00 B5 F3 CF FF F3 CF 3C F7 01 00 00 00 00 00 (74)
[16:58:50][I][Panel :964]: [2025-08-27 16:58] 45: 45 00 9F 39 E7 DF 59 E7 9E FF 01 00 00 00 00 00 (74)
[16:58:50][I][Panel :964]: [2025-08-27 16:58] 71: 71 00 2F EF BE FB EF BE FB 03 00 00 00 00 00 00 (67)
[16:58:50][I][Module:964]: [2025-08-27 16:58] 71: FF 01 0F FF FF FF FF FF FF 03 00 00 00 00 00 00 (67)
[16:58:50][I][Panel :964]: [2025-08-27 16:58] 7D: 7D 00 85 7F 39 E7 9E FF E7 DF 01 00 00 00 00 00 (74)
[16:58:50][I][Panel :964]: [2025-08-27 16:58] 7D: 7D 00 87 E7 3C F7 CF FF F3 CF 01 00 00 00 00 00 (74)
[16:58:50][I][Panel :964]: [2025-08-27 16:58] 05: 05 00 FD E7 9E FF E7 DF 79 E7 01 00 00 00 00 00 (74)
[16:58:50][I][dscalarm:2067]: status E7, last status 01,selection 01,partition=1,skip=0,force=0
[16:58:50][I][Panel :964]: [2025-08-27 16:58] 05: 05 00 87 CD 79 E7 03 DF 39 E7 01 00 00 00 00 00 (74)
[16:58:50][I][dscalarm:2067]: status CD, last status E7,selection 01,partition=1,skip=0,force=0
[16:58:50][I][dscalarm:2067]: status E7, last status FF,selection A5,partition=2,skip=0,force=0
[16:58:50][I][Panel :964]: [2025-08-27 16:58] 05: 05 00 85 3F B3 E7 24 F7 CF FF 01 00 00 00 00 00 (74)
[16:58:50][I][dscalarm:2067]: status 3F, last status CD,selection 01,partition=1,skip=0,force=0
[16:58:50][D][text_sensor:069]: 'line2 Partition 1': Sending state '63'
[16:58:50][I][dscalarm:2067]: status F7, last status DF,selection A5,partition=3,skip=0,force=0
[16:58:50][I][dscalarm:2067]: status FF, last status E7,selection A5,partition=4,skip=0,force=0
[16:58:50][D][text_sensor:069]: 'Partition 1 Msg': Sending state '3f Unknown'
[16:58:50][D][binary_sensor:026]: 'Partition 1 Ready': New state is OFF
[16:58:50][I][Panel :964]: [2025-08-27 16:58] 75: 75 00 85 01 00 C7 79 CF 3C F7 01 00 00 00 00 00 (74)
[16:58:50][I][Panel :964]: [2025-08-27 16:58] 45: 45 00 9F 39 00 DF 79 E7 9E FF 01 00 00 00 00 00 (74)
[16:58:50][I][Panel :964]: [2025-08-27 16:58] 65: 65 00 8F 9D F3 CF 3C F7 CF FF 01 00 00 00 00 00 (74)
[16:58:50][I][Panel :964]: [2025-08-27 16:58] 75: 75 00 87 CF 79 E7 9E FF E7 DF 01 00 00 00 00 00 (74)
[... cut ...]
[16:58:58][I][Panel :964]: [2025-08-27 16:58] 07: 07 00 FF F7 DF FF F7 DF 7D F7 01 00 00 00 00 00 (74)
[16:58:58][I][Panel :964]: [2025-08-27 16:58] 45: 45 00 BF 7B EF FF FB EF BE FF 01 00 00 00 00 00 (74)
[16:58:58][I][Panel :964]: [2025-08-27 16:58] 65: 65 01 9F BD F5 DF 7D F7 1F FF 01 00 00 00 00 00 (74)
[16:58:58][I][Panel :964]: [2025-08-27 16:58] 75: 75 00 AF DF FB EF BE FF EF FF 01 00 00 00 00 00 (74)
[16:58:58][I][Panel :964]: [2025-08-27 16:58] 7D: 7D 00 C7 EF 7D F7 DF FF F7 DF 01 00 00 00 00 00 (74)

abishur, 2025-08-27:
I was able to disconnect the write pin here are my logs:

```
INFO Upload took 10.29 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from *using esphome API
INFO Successfully resolved dscalarm @ * in 0.000s
INFO Successfully connected to dscalarm @ * in 5.193s
INFO Successful handshake with dscalarm @ 192.168.15.198 in 0.142s
[10:15:17][I][app:200]: ESPHome version 2025.8.1 compiled on Aug 26 2025, 15:43:06
[10:15:17][C][wifi:659]: WiFi:
[10:15:17][C][wifi:442]:   Local MAC: *
[10:15:17][C][wifi:447]:   SSID: [redacted]
[10:15:17][C][wifi:450]:   IP Address: *
[10:15:17][C][wifi:454]:   BSSID: [redacted]
[10:15:17][C][wifi:454]:   Hostname: 'dscalarm'
[10:15:17][C][wifi:454]:   Signal strength: -55 dB ▂▄▆█
[10:15:17][C][wifi:465]:   Channel: 1
[10:15:17][C][wifi:465]:   Subnet: *
[10:15:17][C][wifi:465]:   Gateway:*
[10:15:17][C][wifi:465]:   DNS1: *
[10:15:17][C][wifi:465]:   DNS2: *
[10:15:17][C][logger:252]: Logger:
[10:15:17][C][logger:252]:   Max Level: DEBUG
[10:15:17][C][logger:252]:   Initial Level: DEBUG
[10:15:17][C][logger:258]:   Log Baud Rate: 115200
[10:15:17][C][logger:258]:   Hardware UART: UART0
[10:15:17][C][logger:265]:   Task Log Buffer Size: 768
[10:15:17][I][Panel :964]: [2025-08-27 10:15] 85: 85 01 F1 E9 B4 C7 FF FF FF D7 01 00 00 00 00 00 (74)
[10:15:17][C][switch.gpio:087]: GPIO Switch 'Power Front Door Tablet'
[10:15:17][C][switch.gpio:087]:   Restore Mode: always OFF
[10:15:17][C][switch.gpio:029]:   Pin: GPIO27
[10:15:17][C][switch.gpio:087]: GPIO Switch 'Power Garage Door Tablet'
[10:15:17][C][switch.gpio:087]:   Restore Mode: always OFF
[10:15:17][C][switch.gpio:029]:   Pin: GPIO25
[10:15:17][C][template.switch:087]: Template Switch 'Connection'
[10:15:17][C][template.switch:087]:   Restore Mode: disabled 
[10:15:17][C][template.switch:095]:   Icon: 'mdi:shield-link-variant'
[10:15:17][C][template.switch:057]:   Optimistic: YES
[10:15:17][I][Panel :964]: [2025-08-27 10:15] FF: FF 01 81 3F FA CF FD DF E2 FF 01 00 00 00 00 00 (74)
[10:15:17][C][restart:087]: Restart Switch 'restart_switch'
[10:15:17][C][restart:087]:   Restore Mode: always OFF
[10:15:17][C][restart:095]:   Icon: 'mdi:restart'
[10:15:17][C][safe_mode.switch:087]: Safe Mode Switch 'Safe Mode'
[10:15:17][C][safe_mode.switch:087]:   Restore Mode: always OFF
[10:15:17][C][safe_mode.switch:095]:   Icon: 'mdi:restart-alert'
[10:15:17][C][sntp:041]: SNTP Time:
[10:15:17][C][sntp:044]:   Server 0: '0.pool.ntp.org'
[10:15:17][C][sntp:044]:   Server 1: '1.pool.ntp.org'
[10:15:17][C][sntp:044]:   Server 2: '2.pool.ntp.org'
[10:15:17][C][esphome.ota:075]: Over-The-Air updates:
[10:15:17][C][esphome.ota:075]:   Address: dscalarm.local:3232
[10:15:17][C][esphome.ota:075]:   Version: 2
[10:15:17][C][esphome.ota:082]:   Password configured
[10:15:17][I][Panel :964]: [2025-08-27 10:15] 57: 57 01 F7 F7 74 FF C1 DF 0D FF 01 00 00 00 00 00 (74)
[10:15:17][C][safe_mode:018]: Safe Mode:
[10:15:17][C][safe_mode:019]:   Boot considered successful after 60 seconds
[10:15:17][C][safe_mode:019]:   Invoke after 10 boot attempts
[10:15:17][C][safe_mode:019]:   Remain for 300 seconds
[10:15:17][C][api:205]: Server:
[10:15:17][C][api:205]:   Address: dscalarm.local:6053
[10:15:17][C][api:210]:   Noise encryption: YES
[10:15:17][C][mdns:124]: mDNS:
[10:15:17][C][mdns:124]:   Hostname: dscalarm
[10:15:17][I][Panel :964]: [2025-08-27 10:15] 35: 35 01 FF E9 97 FF F7 FF FF FF 01 00 00 00 00 00 (74)
[10:15:17][I][Panel :964]: [2025-08-27 10:15] FD: FD 00 BB FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[10:15:17][I][Panel :964]: [2025-08-27 10:15] 8D: 8D 01 DF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[10:15:17][I][Panel :964]: [2025-08-27 10:15] FD: FD 01 B7 FF FF FF FD FF FF FF 01 00 00 00 00 00 (74)
[10:15:17][I][Panel :964]: [2025-08-27 10:15] FF: FF 00 8D FF F7 FF FF FF FF F7 01 00 00 00 00 00 (74)
[... 41 lines cut ...]
[10:15:21][I][Panel :964]: [2025-08-27 10:15] 9F: 9F 01 BF BF FD F7 EF FF 7D FF 01 00 00 00 00 00 (74)
[10:15:21][I][Panel :964]: [2025-08-27 10:15] F7: F7 00 F7 F7 DF FF FD F7 EF FF 01 00 00 00 00 00 (74)
[10:15:21][I][Panel :964]: [2025-08-27 10:15] 7F: 7F 00 DF FF FB F7 DF FF FD F7 01 00 00 00 00 00 (74)
[10:15:21][I][Panel :964]: [2025-08-27 10:15] E7: E7 00 EF EF BF FF FB EF DF FF 01 00 00 00 00 00 (74)
[10:15:21][I][Panel :964]: [2025-08-27 10:15] FD: FD 00 BD FD F7 DF BE FF F7 EF 01 00 00 00 00 00 (74)
[10:15:21][I][Panel :964]: [2025-08-27 10:15] CD: CD 01 DF DF 7E FF F7 DF BE FF 01 00 00 00 00 00 (74)
[10:15:21][I][Panel :964]: [2025-08-27 10:15] FD: FD 01 BB FD EF DF 7E FF F7 DF 01 00 00 00 00 00 (74)
[10:15:21][I][Panel :964]: [2025-08-27 10:15] C3: C3 01 7F FF FB 01 00 00 00 00 00 00 00 00 00 00 (34)
[10:15:21][I][Panel :964]: [2025-08-27 10:15] F7: F7 00 F7 F7 DF FF FD F7 EF FF 01 00 00 00 00 00 (74)
[10:15:21][I][Panel :964]: [2025-08-27 10:15] 7F: 7F 00 CF FF FB F7 DF FF FD F7 01 00 00 00 00 00 (74)
[10:15:22][I][Panel :964]: [2025-08-27 10:15] E7: E7 00 EF EF BF FF FB EF DF FF 01 00 00 00 00 00 (74)
[10:15:22][I][Panel :964]: [2025-08-27 10:15] FD: FD 00 CD FD F7 EF BF FF FB EF 01 00 00 00 00 00 (74)
[10:15:22][I][Panel :964]: [2025-08-27 10:15] C5: C5 01 DF DF 7E FF F7 DF BE FF 01 00 00 00 00 00 (74)
[10:15:22][I][Panel :964]: [2025-08-27 10:15] FD: FD 01 BB FB FF FF 7D FF EF DF 01 00 00 00 00 00 (74)
[10:15:22][I][dscalarm:2067]: status FF, last status FF,selection 01,partition=1,skip=0,force=1
[... cut ...]
[10:15:22][D][text_sensor:069]: 'Zone Status': Sending state ''
[10:15:22][D][text_sensor:069]: 'Trouble Msg': Sending state ''
[10:15:22][I][Panel :964]: [2025-08-27 10:15] 8F: 8F 01 FF 83 FF C7 07 DF FB DF 01 00 00 00 00 00 (74)
[10:15:22][I][Panel :964]: [2025-08-27 10:15] AD: AD 01 BF A3 BF FF FD F7 EF FF 01 00 00 00 00 00 (74)
[10:15:22][I][Panel :964]: [2025-08-27 10:15] 7F: 7F 01 CF FF FB F7 DF FF FD F7 01 00 00 00 00 00 (74)
[10:15:22][I][Panel :964]: [2025-08-27 10:15] EF: EF 00 F7 F7 DF FF FD F7 EF FF 01 00 00 00 00 00 (74)
```

Dilbert66, 2025-08-27:
Honestly, I can't duplicate the issue you have.  The only thing that makes senses is a wiring on configuration issue with the pins.  The code has been tested on a multitude of esp32's including d1 minis and c3 devices as well as with different interface types.

Dilbert66, 2025-08-28:
Its not necessaraly the the fact that the esp32s2 has limitations but that it might work differently with the arduino libraries.  The latest Arduino and esp_idf versions changed the way the timers work and are initialized (used to time when to read data bits on the bus).  I was using the arduino default esp32 timers but perhaps the library on those chips has some issues with those.  What I've added is a config option to allow swithing to using the base esp_idf timers and bypassing the arduino ones.  
To use them add this config option to the dsc_alarm_panel yaml config section:

```
  use_esp_idf_timer: true
```

abishur, 2025-08-28:
Sadly my other ESP32 still has the same issues even with the use_esp_idf_timer: true option set and using the dev branch for components.  I'll try getting a breadboard set up to use level shifters and break out the lines instead of duplexing off of the one green wire.  Maybe my 4N35 went bad around the time I was doing an updated anyway?  Maybe something about the new timing clocks doesn't gel well with the PC1616 board?  Here are my log files from the different ESP32 and esp_idf_timer set to true:

```
[11:03:21][I][Panel :964]: [2025-08-28 11:03] 05: 05 01 FF CF FF F7 FD FF D6 E7 01 00 00 00 00 00 (74)
[11:03:21][I][dscalarm:2067]: status CF, last status 01,selection 01,partition=1,skip=0,force=0
[11:03:21][I][dscalarm:2067]: status F7, last status C5,selection A3,partition=2,skip=0,force=0
[11:03:21][I][dscalarm:2067]: status E7, last status FF,selection 20,partition=4,skip=0,force=0
[11:03:21][C][safe_mode:018]: Safe Mode:
[11:03:21][C][safe_mode:019]:   Boot considered successful after 60 seconds
[11:03:21][C][safe_mode:019]:   Invoke after 10 boot attempts
[11:03:21][C][safe_mode:019]:   Remain for 300 seconds
[11:03:21][C][api:205]: Server:
[11:03:21][C][api:205]:   Address: dscalarm.local:6053
[11:03:21][C][api:210]:   Noise encryption: YES
[11:03:21][I][Panel :964]: [2025-08-28 11:03] 0D: 0D 01 F1 DF FB FF FF E7 DF FF 01 00 00 00 00 00 (74)
[11:03:21][C][mdns:124]: mDNS:
[11:03:21][C][mdns:124]:   Hostname: dscalarm
[11:03:21][I][Panel :964]: [2025-08-28 11:03] B7: B7 00 B3 1D FF FF FD FF FF FF 01 00 00 00 00 00 (74)
[11:03:21][I][Panel :964]: [2025-08-28 11:03] FD: FD 01 FF EF FE FF DF FF FF DF 01 00 00 00 00 00 (74)
[11:03:21][I][Panel :964]: [2025-08-28 11:03] 15: 15 01 BF FB FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:22][I][Panel :964]: [2025-08-28 11:03] F7: F7 01 FF EF FF FF DF FF FF FF 01 00 00 00 00 00 (74)
[11:03:22][I][Panel :964]: [2025-08-28 11:03] 15: 15 01 FF FF FF FF 7F FF FF FF 01 00 00 00 00 00 (74)
[11:03:22][I][Panel :964]: [2025-08-28 11:03] FD: FD 01 DF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:22][I][Panel :964]: [2025-08-28 11:03] F7: F7 00 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:22][I][Panel :964]: [2025-08-28 11:03] FD: FD 01 BF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:22][I][Panel :964]: [2025-08-28 11:03] CD: CD 01 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:22][I][Panel :964]: [2025-08-28 11:03] F7: F7 00 F7 FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:22][I][Panel :964]: [2025-08-28 11:03] 1F: 1F 01 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:22][I][Panel :964]: [2025-08-28 11:03] F7: F7 00 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:22][I][Panel :964]: [2025-08-28 11:03] FD: FD 01 BF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:22][I][Panel :964]: [2025-08-28 11:03] E5: E5 00 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:22][I][Panel :964]: [2025-08-28 11:03] FD: FD 01 BF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:22][I][Panel :964]: [2025-08-28 11:03] CD: CD 01 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:23][I][Panel :964]: [2025-08-28 11:03] F7: F7 00 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:23][I][Panel :964]: [2025-08-28 11:03] FF: FF 01 BF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:23][I][Panel :964]: [2025-08-28 11:03] E7: E7 01 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:23][I][Panel :964]: [2025-08-28 11:03] DF: DF 01 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:23][I][Panel :964]: [2025-08-28 11:03] F7: F7 00 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:23][I][Panel :964]: [2025-08-28 11:03] DF: DF 01 DF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:23][I][Panel :964]: [2025-08-28 11:03] FD: FD 00 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:23][I][Panel :964]: [2025-08-28 11:03] DF: DF 01 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:23][I][Panel :964]: [2025-08-28 11:03] F7: F7 00 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:23][I][Panel :964]: [2025-08-28 11:03] D7: D7 01 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:23][I][Panel :964]: [2025-08-28 11:03] E7: E7 01 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:23][I][Panel :964]: [2025-08-28 11:03] FD: FD 00 BF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:24][I][Panel :964]: [2025-08-28 11:03] 97: 97 01 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:24][I][Panel :964]: [2025-08-28 11:03] F7: F7 00 BF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:24][I][Panel :964]: [2025-08-28 11:03] 97: 97 01 BF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:24][I][Panel :964]: [2025-08-28 11:03] F7: F7 00 F7 FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:24][I][Panel :964]: [2025-08-28 11:03] 97: 97 01 FF FD FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:24][I][Panel :964]: [2025-08-28 11:03] FD: FD 01 B7 FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:24][I][Panel :964]: [2025-08-28 11:03] 9F: 9F 01 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:24][I][Panel :964]: [2025-08-28 11:03] E5: E5 01 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:24][I][Panel :964]: [2025-08-28 11:03] FD: FD 01 BD FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:24][I][Panel :964]: [2025-08-28 11:03] E7: E7 01 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:24][I][Panel :964]: [2025-08-28 11:03] F7: F7 01 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[... cut ...]
[11:03:52][I][Panel :964]: [2025-08-28 11:03] E5: E5 01 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[11:03:52][I][dscalarm:803]: Writing keys: N to partition 1, partition disabled: 0 , partition locked: 1
[11:03:52][D][text_sensor:069]: 'line1 partition 1': Sending state 'System'
[11:03:52][D][text_sensor:069]: 'line2 partition 1': Sending state 'not available'
[11:03:52][I][Panel :964]: [2025-08-28 11:03] FD: FD 01 BD FF FF FF DF FF D7 DF 01 00 00 00 00 00 (74)
[11:03:52][I][Panel :964]: [2025-08-28 11:03] CD: CD 01 FF FF FF FF FF FF FF EF 01 00 00 00 00 00 (74)
```

Dilbert66, 2025-08-28:
Can you try something else in case you havent yet.  Give it a new name in your yaml so that it will create a whole new directory and compile like new.  Just in case some old code is left over from the previous version.  Yes, I'm grasping here but it's a puzzler.

Dilbert66, 2025-08-29:
Trying out a different circuit is a good idea as I use the same esp32 d1 mini and have no issues with it.

abishur, 2025-09-03:
> I want to confirm that I am experiencing the same issue discussed above. I have reverted back to ESPHome 2025.6.3 and github://Dilbert66/[esphome-components@main-pre2025.7.x](mailto:esphome-components@main-pre2025.7.x) and my sensor readings have come back.

What setup are you using to communicate with the dsc panel?  I've been using an ESP32 and trio of 4n35 optocouplers and duplexing the green line for comms.

Dilbert66, 2025-09-03:
chtran46, glad you got it sorted out. tks for confirming.

Dilbert66, 2025-09-04:
Can you post some logs when sending a cmd so we can see what is happening. You should see a 05 panel packet every time you send a cmd

abishur, 2025-09-04:
```
[10:48:58][I][app:200]: ESPHome version 2025.8.3 compiled on Sep  4 2025, 08:35:30
[10:48:58][C][wifi:661]: WiFi:
[10:48:58][C][wifi:444]:   Local MAC: 08:D1:F9:27:ED:10
[10:48:58][C][wifi:449]:   SSID: [redacted]
[10:48:58][C][wifi:452]:   IP Address: *.*.*.*
[10:48:59][C][wifi:456]:   BSSID: [redacted]
[10:48:59][C][wifi:456]:   Hostname: 'dscalarm'
[10:48:59][C][wifi:456]:   Signal strength: -56 dB ▂▄▆█
[10:48:59][C][wifi:467]:   Channel: 1
[10:48:59][C][wifi:467]:   Subnet: 255.255.254.0
[10:48:59][C][wifi:467]:   Gateway: *.*.*.*
[10:48:59][C][wifi:467]:   DNS1: *.*.*.*
[10:48:59][C][wifi:467]:   DNS2: *.*.*.*
[10:48:59][C][logger:252]: Logger:
[10:48:59][C][logger:252]:   Max Level: DEBUG
[10:48:59][C][logger:252]:   Initial Level: DEBUG
[10:48:59][C][logger:258]:   Log Baud Rate: 115200
[10:48:59][C][logger:258]:   Hardware UART: UART0
[10:48:59][C][logger:265]:   Task Log Buffer Size: 768
[10:48:59][C][switch.gpio:087]: GPIO Switch 'Power Front Door Tablet'
[10:48:59][C][switch.gpio:087]:   Restore Mode: always OFF
[10:48:59][C][switch.gpio:029]:   Pin: GPIO27
[10:48:59][C][switch.gpio:087]: GPIO Switch 'Power Garage Door Tablet'
[10:48:59][C][switch.gpio:087]:   Restore Mode: always OFF
[10:48:59][C][switch.gpio:029]:   Pin: GPIO25
[10:48:59][C][template.switch:087]: Template Switch 'Connection'
[10:48:59][C][template.switch:087]:   Restore Mode: disabled 
[10:48:59][C][template.switch:095]:   Icon: 'mdi:shield-link-variant'
[10:48:59][C][template.switch:057]:   Optimistic: YES
[10:48:59][C][restart:087]: Restart Switch 'restart_switch'
[10:48:59][C][restart:087]:   Restore Mode: always OFF
[10:48:59][C][restart:095]:   Icon: 'mdi:restart'
[10:48:59][C][safe_mode.switch:087]: Safe Mode Switch 'Safe Mode'
[10:48:59][C][safe_mode.switch:087]:   Restore Mode: always OFF
[10:48:59][C][safe_mode.switch:095]:   Icon: 'mdi:restart-alert'
[10:48:59][C][sntp:041]: SNTP Time:
[10:48:59][C][sntp:044]:   Server 0: '0.pool.ntp.org'
[10:48:59][C][sntp:044]:   Server 1: '1.pool.ntp.org'
[10:48:59][C][sntp:044]:   Server 2: '2.pool.ntp.org'
[10:48:59][C][esphome.ota:075]: Over-The-Air updates:
[10:48:59][C][esphome.ota:075]:   Address: dscalarm.local:3232
[10:48:59][C][esphome.ota:075]:   Version: 2
[10:48:59][C][esphome.ota:082]:   Password configured
[10:48:59][C][safe_mode:018]: Safe Mode:
[10:48:59][C][safe_mode:019]:   Boot considered successful after 60 seconds
[10:48:59][C][safe_mode:019]:   Invoke after 10 boot attempts
[10:48:59][C][safe_mode:019]:   Remain for 300 seconds
[10:48:59][C][api:205]: Server:
[10:48:59][C][api:205]:   Address: dscalarm.local:6053
[10:48:59][C][api:210]:   Noise encryption: YES
[10:48:59][C][mdns:124]: mDNS:
[10:48:59][C][mdns:124]:   Hostname: dscalarm
[10:49:00][I][Panel :964]: [2025-09-04 10:49] 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00 (67)
[10:49:15][I][Panel :964]: [2025-09-04 10:49] B1: B1 00 07 FF FF 03 00 00 00 00 B9 00 00 00 00 00 (82)
[10:49:17][I][dscalarm:803]: Writing keys: N to partition 1, partition disabled: 0 , partition locked: 0
[10:49:17][I][dscalarm:2071]: status 01, last status 01,selection 01,partition=1,skip=0,force=1
[10:49:23][I][Panel :964]: [2025-09-04 10:49] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
```

Dilbert66, 2025-09-04:
Is your write pin connected? It looks like an issue with your write line.

abishur, 2025-09-04:
I'll double check the write pin when I get back tonight.

For duplexing, I thought you were referring to the way that the SDA and CS pins both used the same "green" wire.  I just looked at the circuit diagram example for the mosfets and see what you mean by duplexing.  I am not doing that, lol.  Sorry for the confusion.  Other than using different pins I am following the example of the optocoupler circuit.

abishur, 2025-09-05:
Everything is working again 🎉

I'm on ESPHome version 2025.8.3 which I noticed in the released notes fixed a few bugs pertaining to IDF stuff, so maybe that's what finally got it working again.

I did still have to use
```
use_esp_idf_timer: true
```
in the dsc_alarm_panel area, without that I still got the log full of junk data.

The reason I wasn't able to write to the panel after the update is because I had changed the write pin as part of some testing I was doing while everything was broken.  Once I had put it back to the correct pin just now it began to write again.

Dilbert66, 2025-09-09:
I'll leave the issue open in case others have the same problem and can see the solution and/or provide new details.  Hard to know what it can be.. Some slight variation in the timers and how the arduino library handles it perhaps.  I'll have a look at the  arduino source and see if I can spot something.

Dilbert66, 2025-11-15:
It looks like on some devices the arduino code selects an incorrect clock source and timing is off.  I've amended the code to now default to using the esp-idf general purpose timers directly with a known config.  That seems to solve the issue.  I've updated both main and dev to the new config.

closed 2025-11-15 by Dilbert66