Dilbert66/esphome-dsckeybus #182: dscalarm Partition 1 Status (ps_1)  ARMING for hours

nicocalvagna, 2025-08-30:
Hi, it's me again. I marked the previous thread as closed because I was able to solve the constant crash problem. I've had it running for days without a single disconnection with the new ESP32.

Unfortunately, the initial problem is still present, and it's not due to the reboots, but rather another factor causing it.

The situation is as follows: when disarming the alarm, the Partition 1 Status changes to disarmed. But then, sometimes in seconds and sometimes in minutes, it changes to arming, and it stays there for hours. This creates the problem of not being able to arm again. I have to arm from the physical keypad, and then it updates correctly. Or if I call the disarm service, with the corresponding code, the system arms. I have to press disarm again and then arm away, for example. This didn't happen before, and the alarm system is exactly the same, so I'm ruling out a fault in the data bus or something else.

It's a DSC 1832 with a PC5132 receiver, wireless zones, and some wired zones on the board.

I've attached a screenshot of the history of this device in question.

[image]
[image]

Any ideas? I'm willing to try whatever it takes.

Thank you very much.

Dilbert66, 2025-08-30:
The home assistant images are not useful for debugging.  I need to see the esphome logs when this is happening.  Do a disarming sequence while capture the logs and post them here. Otherwise, I have no idea what your isssue is.

nicocalvagna, 2025-08-30:
the "arming" status ocurr at 11.22hs. here is a fragment of that log.

[11:22:00][I][Panel :964]: [2025-08-30 11:22] 05: 05 00 88 08 06 38 06 38 06 0F 00 76 00 00 00 00 (71)
[11:22:00][I][dscalarm:2067]: status 08, last status 01,selection 01,partition=1,skip=0,force=0
[11:22:00][D][text_sensor:069]: 'line1 Partition 1 (ln1_1)': Sending state 'Exit delay'
[11:22:00][D][text_sensor:069]: 'line2 Partition 1 (ln2_1)': Sending state 'in progress'
[11:22:00][I][dscalarm:2067]: status 38, last status F8,selection 00,partition=2,skip=0,force=0
[11:22:00][D][text_sensor:069]: 'Partition 1 Msg (msg_1)': Sending state '08 Exit delay'
[11:22:00][D][text_sensor:069]: 'Partition 1 Status (ps_1)': Sending state 'arming'
[11:22:00][I][Panel :964]: [2025-08-30 11:22] 05: 05 00 81 01 00 C7 00 C7 00 C7 01 00 00 00 00 00 (74)
[11:22:00][I][dscalarm:2067]: status 01, last status 08,selection 01,partition=1,skip=0,force=0
[11:22:00][D][text_sensor:069]: 'line1 Partition 1 (ln1_1)': Sending state 'System is Ready'
[11:22:00][D][text_sensor:069]: 'line2 Partition 1 (ln2_1)': Sending state 'Ready to Arm <>'
[11:22:00][D][text_sensor:069]: 'Partition 1 Msg (msg_1)': Sending state '01 Ready'
[11:22:01][I][Panel :964]: [2025-08-30 11:22] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[11:22:04][I][Panel :964]: [2025-08-30 11:22] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[11:22:07][I][Panel :964]: [2025-08-30 11:22] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[11:22:10][I][Panel :964]: [2025-08-30 11:22] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[11:22:13][I][Panel :964]: [2025-08-30 11:22] E6: E6 00 2C 08 00 00 00 00 1A 00 00 00 00 00 00 00 (66)
[11:22:13][I][Panel :964]: [2025-08-30 11:22] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)

nicocalvagna, 2025-08-30:
the circuit is the one whith resistors. i have cleaned the build files in esphome and compiled again with no luck, the same problem.

here is the log.

[13:52:18][I][Panel :964]: [2025-08-30 13:52] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[13:52:19][I][Panel :964]: [2025-08-30 13:52] 05: 05 00 81 01 00 C7 00 C7 00 C7 01 00 00 00 00 00 (74)
[13:52:19][I][dscalarm:2071]: status 01, last status 3E,selection 01,partition=1,skip=0,force=0
[13:52:19][D][text_sensor:069]: 'line1 Partition 1 (ln1_1)': Sending state 'System is Ready'
[13:52:19][D][text_sensor:069]: 'line2 Partition 1 (ln2_1)': Sending state 'Ready to Arm <>'
[13:52:19][D][text_sensor:069]: 'Partition 1 Msg (msg_1)': Sending state '01 Ready'
[13:52:19][D][dht:042]: Temperature 19.0°C Humidity 72.0%
[13:52:19][D][sensor:103]: 'Temperatura Living': Sending state 19.00000 °C with 1 decimals of accuracy
[13:52:19][D][sensor:103]: 'Humedad Living': Sending state 72.00000 % with 0 decimals of accuracy
[13:52:21][I][Panel :964]: [2025-08-30 13:52] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[13:52:22][I][Panel :964]: [2025-08-30 13:52] 27: 27 00 81 01 00 C7 04 74 00 00 00 00 00 00 00 00 (58)
[13:52:22][D][binary_sensor:026]: 'Zona 3': New state is ON
[13:52:22][D][text_sensor:069]: 'zone status (zs)': Sending state 'OP:3'
[13:52:22][I][Panel :964]: [2025-08-30 13:52] 05: 05 00 81 02 00 C7 00 C7 00 C7 01 00 00 00 00 00 (74)
[13:52:22][I][dscalarm:2071]: status 02, last status 01,selection 01,partition=1,skip=0,force=0
[13:52:22][D][text_sensor:069]: 'Partition 1 Msg (msg_1)': Sending state '02 Stay zones open'
[13:52:23][I][Panel :964]: [2025-08-30 13:52] 27: 27 00 81 02 00 C7 00 71 00 00 00 00 00 00 00 00 (58)
[13:52:23][D][binary_sensor:026]: 'Zona 3': New state is OFF
[13:52:23][D][text_sensor:069]: 'zone status (zs)': Sending state ''
[13:52:24][I][Panel :964]: [2025-08-30 13:52] 05: 05 00 81 01 00 C7 00 C7 00 C7 01 00 00 00 00 00 (74)
[13:52:24][I][dscalarm:2071]: status 01, last status 02,selection 01,partition=1,skip=0,force=0
[13:52:24][D][text_sensor:069]: 'Partition 1 Msg (msg_1)': Sending state '01 Ready'
[13:52:24][I][Panel :964]: [2025-08-30 13:52] 27: 27 00 81 01 00 C7 04 74 00 00 00 00 00 00 00 00 (58)
[13:52:24][D][binary_sensor:026]: 'Zona 3': New state is ON
[13:52:24][D][text_sensor:069]: 'zone status (zs)': Sending state 'OP:3'
[13:52:24][I][Panel :964]: [2025-08-30 13:52] 05: 05 00 81 02 00 C7 00 C7 00 C7 01 00 00 00 00 00 (74)
[13:52:24][I][dscalarm:2071]: status 02, last status 01,selection 01,partition=1,skip=0,force=0
[13:52:24][D][text_sensor:069]: 'Partition 1 Msg (msg_1)': Sending state '02 Stay zones open'
[13:52:24][I][Panel :964]: [2025-08-30 13:52] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[13:52:26][I][Panel :964]: [2025-08-30 13:52] 27: 27 00 81 02 00 C7 00 71 00 00 00 00 00 00 00 00 (58)
[13:52:26][D][binary_sensor:026]: 'Zona 3': New state is OFF
[13:52:26][D][text_sensor:069]: 'zone status (zs)': Sending state ''
[13:52:26][I][Panel :964]: [2025-08-30 13:52] 05: 05 00 81 01 00 C7 00 C7 00 C7 01 00 00 00 00 00 (74)
[13:52:26][I][dscalarm:2071]: status 01, last status 02,selection 01,partition=1,skip=0,force=0
[13:52:26][D][text_sensor:069]: 'Partition 1 Msg (msg_1)': Sending state '01 Ready'
[13:52:27][I][Panel :964]: [2025-08-30 13:52] 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00 (67)
[13:52:27][I][Module:964]: [2025-08-30 13:52] 11: FF 01 3F FF FF F3 FF FF FF 03 00 00 00 00 00 00 (67)
[13:52:27][I][Panel :964]: [2025-08-30 13:52] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[13:52:27][I][Panel :964]: [2025-08-30 13:52] 27: 27 00 81 01 00 C7 04 74 00 00 00 00 00 00 00 00 (58)
[13:52:27][D][binary_sensor:026]: 'Zona 3': New state is ON
[13:52:27][D][text_sensor:069]: 'zone status (zs)': Sending state 'OP:3'
[13:52:28][I][Panel :964]: [2025-08-30 13:52] 05: 05 00 81 02 00 C7 00 C7 00 C7 01 00 00 00 00 00 (74)
[13:52:28][I][dscalarm:2071]: status 02, last status 01,selection 01,partition=1,skip=0,force=0
[13:52:28][D][text_sensor:069]: 'Partition 1 Msg (msg_1)': Sending state '02 Stay zones open'
[13:52:29][I][Panel :964]: [2025-08-30 13:52] 27: 27 00 81 02 00 C7 00 71 00 00 00 00 00 00 00 00 (58)
[13:52:29][D][binary_sensor:026]: 'Zona 3': New state is OFF
[13:52:29][D][text_sensor:069]: 'zone status (zs)': Sending state ''
[13:52:29][I][Panel :964]: [2025-08-30 13:52] 05: 05 00 81 01 00 C7 00 C7 00 C7 01 00 00 00 00 00 (74)
[13:52:29][I][dscalarm:2071]: status 01, last status 02,selection 01,partition=1,skip=0,force=0
[13:52:29][D][text_sensor:069]: 'Partition 1 Msg (msg_1)': Sending state '01 Ready'
[13:52:30][I][Panel :964]: [2025-08-30 13:52] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[13:52:33][I][Panel :964]: [2025-08-30 13:52] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[13:52:36][I][Panel :964]: [2025-08-30 13:52] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[13:52:39][I][Panel :964]: [2025-08-30 13:52] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[13:52:42][I][Panel :964]: [2025-08-30 13:52] 16: 16 00 0E 46 F5 5F 00 00 00 00 00 00 00 00 00 00 (42)
[13:52:42][I][Panel :964]: [2025-08-30 13:52] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[13:52:45][I][Panel :964]: [2025-08-30 13:52] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[13:52:48][I][Panel :964]: [2025-08-30 13:52] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[... cut ...]
[14:03:33][I][Panel :964]: [2025-08-30 14:03] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:03:36][I][Panel :964]: [2025-08-30 14:03] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:03:39][I][Panel :964]: [2025-08-30 14:03] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:03:42][I][Panel :964]: [2025-08-30 14:03] A5: A5 00 25 23 CE 10 00 00 CB 00 00 00 00 00 00 00 (66)
[14:03:42][I][Panel :964]: [2025-08-30 14:03] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)

Dilbert66, 2025-08-30:
What do you mean no luck? I dont' see the arming message anywhere in the log?  The status are not changing like they were. 
I don't see any issues there.  Yes, you do have some bit errrors but they are not affecting the status.

nicocalvagna, 2025-08-30:
im sorry, accidentally copy the log incomplete, here is the missing part, where at 14:04hs the state is "arming".

[14:01:19][I][Panel :964]: [2025-08-30 14:01] 01: 01 01 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 (16)
[14:01:21][I][Panel :964]: [2025-08-30 14:01] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:01:24][I][Panel :964]: [2025-08-30 14:01] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:01:27][I][Panel :964]: [2025-08-30 14:01] 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00 (67)
[14:01:27][I][Module:964]: [2025-08-30 14:01] 11: FF 01 3F FF FF F3 FF FF FF 03 00 00 00 00 00 00 (67)
[14:01:27][I][Panel :964]: [2025-08-30 14:01] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:01:30][I][Panel :964]: [2025-08-30 14:01] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:01:33][I][Panel :964]: [2025-08-30 14:01] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:01:36][I][Panel :964]: [2025-08-30 14:01] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:01:39][I][Panel :964]: [2025-08-30 14:01] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:01:42][I][Panel :964]: [2025-08-30 14:01] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:01:42][I][Panel :964]: [2025-08-30 14:01] B1: B1 00 FF FD FF 00 00 00 00 00 AC 00 00 00 00 00 (82)
[14:01:45][I][Panel :964]: [2025-08-30 14:01] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:01:48][I][Panel :964]: [2025-08-30 14:01] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:01:51][I][Panel :964]: [2025-08-30 14:01] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:01:54][I][Panel :964]: [2025-08-30 14:01] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:01:57][I][Panel :964]: [2025-08-30 14:01] 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00 (67)
[14:01:57][I][Module:964]: [2025-08-30 14:01] 11: FF 01 3F FF FF F3 FF FF FF 03 00 00 00 00 00 00 (67)
[14:01:57][I][Panel :964]: [2025-08-30 14:01] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:02:00][I][Panel :964]: [2025-08-30 14:02] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:02:03][I][Panel :964]: [2025-08-30 14:02] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:02:06][I][Panel :964]: [2025-08-30 14:02] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:02:09][I][Panel :964]: [2025-08-30 14:02] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:02:12][I][Panel :964]: [2025-08-30 14:02] E6: E6 00 2C 08 00 00 00 00 1A 00 00 00 00 00 00 00 (66)
[14:02:12][I][Panel :964]: [2025-08-30 14:02] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:02:15][I][Panel :964]: [2025-08-30 14:02] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:02:16][I][Panel :964]: [2025-08-30 14:02] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:02:16][I][dscalarm:2071]: status 01, last status 01,selection 01,partition=1,skip=0,force=1
[14:02:18][D][dht:042]: Temperature 20.0°C Humidity 18.0%
[14:02:18][D][sensor:103]: 'Temperatura Pieza': Sending state 20.00000 °C with 1 decimals of accuracy
[14:02:18][D][sensor:103]: 'Humedad Pieza': Sending state 18.00000 % with 0 decimals of accuracy
[14:02:18][I][Panel :964]: [2025-08-30 14:02] 05: 05 00 81 01 00 C7 00 C0 06 0F 00 77 00 00 00 00 (71)
[14:02:18][W][dscalarm:1580]: Bit count mismatch. Ignoring cmd
[14:02:18][I][dscalarm:2071]: status C0, last status 38,selection 00,partition=3,skip=0,force=0
[14:02:18][I][Panel :964]: [2025-08-30 14:02] 05: 05 00 81 01 00 C7 00 C7 00 C7 01 00 00 00 00 00 (74)
[14:02:18][I][Panel :964]: [2025-08-30 14:02] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:02:19][D][dht:042]: Temperature 19.0°C Humidity 72.0%
[14:02:19][D][sensor:103]: 'Temperatura Living': Sending state 19.00000 °C with 1 decimals of accuracy
[14:02:19][D][sensor:103]: 'Humedad Living': Sending state 72.00000 % with 0 decimals of accuracy
[14:02:19][I][Panel :964]: [2025-08-30 14:02] C7: C7 00 01 8E 01 8F 00 00 00 00 00 00 00 00 00 00 (41)
[14:02:21][I][Panel :964]: [2025-08-30 14:02] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:02:24][I][Panel :964]: [2025-08-30 14:02] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:02:27][I][Panel :964]: [2025-08-30 14:02] 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00 (67)
[14:02:27][I][Module:964]: [2025-08-30 14:02] 11: FF 01 3F FF FF F3 FF FF FF 03 00 00 00 00 00 00 (67)
[14:02:27][I][Panel :964]: [2025-08-30 14:02] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:02:30][I][Panel :964]: [2025-08-30 14:02] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:02:33][I][Panel :964]: [2025-08-30 14:02] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:02:36][I][Panel :964]: [2025-08-30 14:02] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:02:39][I][Panel :964]: [2025-08-30 14:02] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:02:42][I][Panel :964]: [2025-08-30 14:02] 3E: 3E 00 81 01 00 C7 00 87 00 00 00 00 00 00 00 00 (58)
[14:02:42][I][Panel :964]: [2025-08-30 14:02] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:02:45][I][Panel :964]: [2025-08-30 14:02] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[14:02:48][I][Panel :964]: [2025-08-30 14:02] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[... cut ...]
[14:05:17][I][Panel :964]: [2025-08-30 14:05] 27: 27 00 81 01 00 C7 04 74 00 00 00 00 00 00 00 00 (58)
[14:05:17][D][binary_sensor:026]: 'Zona 3': New state is ON
[14:05:17][D][text_sensor:069]: 'zone status (zs)': Sending state 'OP:3'
[14:05:17][I][Panel :964]: [2025-08-30 14:05] 05: 05 00 81 02 00 C7 00 C7 00 C7 01 00 00 00 00 00 (74)
[14:05:17][I][dscalarm:2071]: status 02, last status 01,selection 01,partition=1,skip=0,force=0
[14:05:17][D][text_sensor:069]: 'Partition 1 Msg (msg_1)': Sending state '02 Stay zones open'

Dilbert66, 2025-08-30:
Ok, i see it. I've made a few code tweaks to filter out bad cmds as well as to ensure the status is reset properly.  Recompile again and try again.

nicocalvagna, 2025-08-30:
I can say after 4 hours that it's working. I'll keep testing and check the Keybus wiring just in case, but so far it hasn't changed to "arming"; it's still in "disarmed" mode while I'm here at home.

Thank you very much!

closed 2025-08-30 by nicocalvagna